scanse/sweep-sdk #16: Improve get_scan throughput

daniel-j-h, 2016-11-12:
At the moment `get_scan` waits for the sync bit then records responses until the next sync bit. This makes sure we gather and report only full 360 degree scans:

https://github.com/scanse/sweep-sdk/blob/853f113543621daf6ce8553177f7d2153d2657c4/libsweep/sweep.c#L177-L231

But it's also wasteful since we're throwing away the data before the first sync bit, limiting our throughput.

To keep the low-level library simple I don't want to go the route of a worker thread, accumulating behind the scenes. Instead, we should be able to check for the angle we start recording at and stop recording as soon as we reach it again.

We don't guarantee for the first sample to start at angle 0 degree after all. If the user needs this, she can accumulate herself.

daniel-j-h, 2016-11-15:
Looks like the Python bindings fail to keep up with polling in some use-cases.

I could see read errors when using threads in Python for polling + clustering + drawing.

This means, we have to put back the threaded worker into `libsweep` for accumulating scan responses.

dcyoung, 2017-01-18:
I'm not sure what you mean by "throughput" here. Could you clarify? Trashing all the data before the first error/sync byte will mean that data collected in-between the receipt header and the first sync byte will go unreported. Functionally it would present to the host as a slight delay before the sensor starts to acquire data. But once the first sync byte is noted, the throughput is consistent yes? 

The number of sensor readings in a scan is variable, so any user application will have to support variable number of data points in a scan. As long as we package them in the same format as complete scans, is there a problem with reporting partial scans? Thinking ahead, if we add a very slow motor speed (<<1Hz), it could be unwise to limit reporting to only complete scans.  I could see some applications wanting a guarantee of a full scan however, so perhaps we should consider a setting that toggles the reporting of partial scans. Ie: if the user sends a "DS" command with the setting enabled, then the first partial scan is included, whereas the partial scan is trashed if the setting is disabled. 

Additionally, for efficiency, we may want to modify `SWEEP_MAX_SAMPLES` based on the device's current settings, given that various combinations of motor speed and sample rate will produce drastically different point counts in a single scan.

daniel-j-h, 2017-01-19:
See [here](https://github.com/scanse/sweep-sdk/blob/853f113543621daf6ce8553177f7d2153d2657c4/libsweep/sweep.c#L177-L231). At the moment we
- wait for the sync bit
- start recording
- wait for another sync bit to indicate the end   <- (1)
- return this as a full scan
- repeat

(1) is the issue here. We re-use the sync bit both for indicating the start but also the end. Therefore we throw away a a full scan for every scan we do. This is ugly and should change.

At the moment we also rely on the user to poll for new scans. When playing with some Python code I experienced the Python interpreter sometimes (GC?) being too slow to keep up with the scans.

To avoid this we could put a worker thread back into the libsweep library polling for scans all the time and aggregating scans. Then the user is able to poll for full scans (in the style of a SPSC queue). This was the initial plan but was reworked without the threading worker for simplicity reasons. cc @kent-williams 

We can of course think about reporting a stream of samples without having the user to know about the sync bit. If users need this they can deduce it based on the angle I guess. We probably still need a worker thread to keep up with the device's stream of data (?) (Maybe even setting a thread scheduler priority?).

dcyoung, 2017-01-25:
Definitely agree. The necessary move is to add a worker thread that aggregates (reads and parses the incoming DATA_BLOCK receipts). 

The implementation of threads has sparked some discussion of infrastructure, so until we're sure what implementation will be used, we can implement a temporary fix to increase throughput. Something like this: https://github.com/dcyoung/sweep-sdk/commit/7af13cd00fb629d773bc37d79253249018a9eb3c

This isn't ideal because you don't guarantee a complete scan. But this is OK for now, because scans are inherently variable.. both in quantity of readings and in the spacing between readings. Therefore any user applications must be able to deal with variability already. With this temporary fix, polling `sweep_device_get_scan` repeatedly will produce the following behavior:
1. First return from `sweep_device_get_scan` will be a partial scan containing all the readings accumulated after the function was called, up until the sync byte is encountered. (ie: you get the data points of the current scan).
2. Subsequent returns from `sweep_device_get_scan` will return scans that are missing their first reading (the sync byte reading). 

These drawbacks are preferable to trashing every other scan in its entirety. Again, the quantity and spacing of readings in a scan is inherently variable, so there is no reason a user application shouldn't be able to handle this return style. If the user really wants to avoid receiving the first partial scan, they can simply trash it themselves and only use subsequent scans. 

Again, this would be a temporary fix until the worker thread is implementation replaces it. Which I aim to jump on ASAP.

dcyoung, 2017-02-02:
A rough version of this in the `win-support` branch....
https://github.com/scanse/sweep-sdk/blob/14679353b247c3909c29966fca3d4b99d67bec7e/libsweep/src/sweep.c#L203-L236

MikeGitb, 2017-02-19:
@dcyoung: If that is OK with you I'd put that approach into a separate PR.

dcyoung, 2017-04-21:
Resolved in b867c37d7dd4f2d1f39dad4819fd1ecae04e143a for the case of the background thread accumulating scans. 

In the future if we add alternative methods that get a scan directly, we may need to re-investigate the issue of parsing that sync byte. 

Closing for now

closed 2017-04-21 by dcyoung