scanse/sweep-sdk #19: High-resolution timestamp for each sample

daniel-j-h, 2016-11-12:
At the moment we accumulate a full 360 degree scan and return this scan to the user.

Many use-cases need high-resolution synchronization which is currently not possible on a per-sample basis to do for the user (could be estimated using motor speed and timestamp on a scan, though).

For usability reasons it would be great to have a timestamp per angle, distance sample.

Tasks:
- [ ] time.h needs a timestamp interface
- [ ] time_unix.c implementation
- [ ] sweep.h needs a timestamp interface for samples (similar to get_distance, get_angle, ..)
- [ ] sweep.c and dummy.c implementations
- [ ] add to sweep.hpp
- [ ] add to sweeppy
- [ ] add to sweepjs

kent-williams, 2016-11-13:
Just FYI, We plan to add a timestamp to each scan packet in the future, but it first requires increasing the serial baud rate to make room for the extra bytes. I think this is a much needed addition though for the time being.

MikeGitb, 2017-02-01:
Will those time stamps be available in the next batch of devices you ship?

MikeGitb, 2017-05-08:
Is attaching a time stamp to each measurement package still of interest? If so, I could provide a PR.