scanse/sweep-sdk #20: Implement error code bit

daniel-j-h, 2017-01-04:
https://github.com/scanse/sweep-sdk/commit/a0d54589daa4cd717b917376a6c2deeff96a143f introduced an error code bit. Currently we only check the sync bit [here](https://github.com/scanse/sweep-sdk/blob/a0d54589daa4cd717b917376a6c2deeff96a143f/libsweep/sweep.c#L200) to check when a new full scan starts. If the error bit is set we should throw away the scan.

kent-williams, 2017-01-04:
If there is an error code received, this only indicates a communication error on that particular ranging and not the complete 360 scan, just to clarify!

dcyoung, 2017-01-19:
This should be high on the priority list, as communication errors are not unheard of.  We should also update the docs for the error/sync byte as it is a bit unclear. Until then, this article explains it a little better:
https://support.scanse.io/hc/en-us/articles/115001403947-Parsing-Sensor-Reading-Data

daniel-j-h, 2017-01-19:
[Here](https://github.com/scanse/sweep-sdk/blob/853f113543621daf6ce8553177f7d2153d2657c4/libsweep/sweep.c#L177-L231) is the location where we handle scan responses.

Before checking the sync bit you can check for the error bit and `continue` the receive loop. Maybe also add a counter and bail out completely after `n` error bits per full scan. Otherwise we could loop forever.

Note: reporting such fatal errors to users is fine - we have proper error handling in place and expect users to handle these (or handle them automatically e.g. by throwing an exception in C++ land).

commit 02f3e955 (2017-02-12): Implements Communication Error Bit, resolves #20
commit 6c3ae36a (2017-02-13): Implements Communication Error Bit, resolves #20
closed 2017-02-13 by daniel-j-h