scanse/sweep-sdk #23: Examples Error on subsequent runs without device reset

dcyoung, 2017-01-19:
Running the examples `./example-c` or ./example-c++` works the first time, but errors on subsequent attempts stating: 

> Error: unable to receive stop scanning command response

Has this always been the behavior or is it a new issue? I haven't started tracking down the origin of the issue yet, but I'm assuming the device is not being shutdown properly.

daniel-j-h, 2017-01-19:
No, this should not happen. I never use the reset command in my examples and playground work (except for testing it once while implementing it). But I also used some older device versions.

The [`sweep_device_stop_scanning`](https://github.com/scanse/sweep-sdk/blob/a33032a5626d43e1d72b3457dda8f6ebb6bc86ca/libsweep/sweep.c#L130-L175) function is the only place where we send the stop command. We
- send the stop command
- wait a bit (kinda ugly) for the device to receive and stop sending data
- send the stop command again and wait for the ack response

When we initialize a new Sweep device in `sweep_device_construct` we also call the `sweep_device_stop_scanning` function [here](https://github.com/scanse/sweep-sdk/blob/a33032a5626d43e1d72b3457dda8f6ebb6bc86ca/libsweep/sweep.c#L83) to work around already started devices.

I remember talking to @kent-williams about this but back then this was what we were supposed to do.

daniel-j-h, 2017-01-22:
Just tried to reproduce this issue with the device I have here (note this is a old version, the new ones will hopefully arrive next week). But I was unable to.

What I tried:
- compile and install `libsweep`
- compile the examples
- plug in the device
- start the viewer example (works), stop the example
- start the viewer example a second time (works), stop the example
- unplug the device
- plug in the device
- start the viewer example (works), stop the example
- start the viewer example a second time (works), stop the example

All the viewer examples does is starting the scanning and receiving responses [here](https://github.com/scanse/sweep-sdk/blob/a33032a5626d43e1d72b3457dda8f6ebb6bc86ca/libsweep/examples/viewer.cc#L63-L69).

As soon as the new units arrive I can test and try to reproduce again. Any hints from your side what could happen here? cc @kent-williams @dcyoung

dcyoung, 2017-01-23:
Odd that the behavior was not reproduce-able. I'll double check on my end again in a bit. 

> I never use the reset command in my examples

Sorry for the confusion... When i said "reset", I was actually referring to resetting the device manually (unplugging it and plugging it back in), not sending the reset command over serial. 

> The `sweep_device_stop_scanning` function is the only place where we send the stop command

I'm aware that the stop command is transmitted in the referenced method, but I'm still unable to find where the method `sweep_device_stop_scanning` is called in the examples? Consider `example.c`. It appears that:
- `sweep_device_start_scanning` method initiates the acquisition
- a scan is read with `sweep_device_get_scan`, and then parsed with other methods (REPEATS 10 times)
- the device is destroyed using  `sweep_device_destruct`

Am I missing where the `sweep_device_stop_scanning` method comes into play? It seems like it would be placed at the end of the acquisition, and before the destruction of the device.

daniel-j-h, 2017-01-25:
> Am I missing where the sweep_device_stop_scanning method comes into play? It seems like it would be placed at the end of the acquisition, and before the destruction of the device.

Good catch, this could be done [here](https://github.com/scanse/sweep-sdk/blob/9fbe17d0946a7a42c7e49644983a5f8e27af74eb/libsweep/examples/example.c#L65-L66) before shutdown.

Not sure if the `sweep_device_destruct` function should stop the device from scanning. I'd like to keep the symmetry as in device construct / destruct, start / stop scanning. But then again we can just call `sweep_device_stop_scanning` in `sweep_device_destruct`:
- in case the user already sent the stop command we exit early [here](https://github.com/scanse/sweep-sdk/blob/9fbe17d0946a7a42c7e49644983a5f8e27af74eb/libsweep/sweep.c#L135-L136) anyway
- in case the user did not send the stop command, we handle it for the user transparently

> @kent-williams was mentioning that the serial port needs to be flushed on destruction, which I don't think we're doing currently.

Oh? This is good to know! If so we should do this in the destruct function [here](https://github.com/scanse/sweep-sdk/blob/9fbe17d0946a7a42c7e49644983a5f8e27af74eb/libsweep/sweep.c#L97) where we still have access to the serial port. Check the serial interface for a flush function.

To summarize:
- [ ] Let `sweep_device_destruct` call `sweep_device_stop_scanning` before destructing the device
- [ ] Flush the serial device before destructing it in `sweep_device_destruct`. Should this be done in [`sweep_serial_device_destruct`](https://github.com/scanse/sweep-sdk/blob/9fbe17d0946a7a42c7e49644983a5f8e27af74eb/libsweep/serial_unix.c#L359-L363)? Strictly speaking flushing the device is the Sweep driver's and not the serial driver's responsibility.
- [ ] Adapt examples (and bindings) to still call `sweep_device_stop_scanning` to show user API symmetry
- [ ] Verify with device and check if this fixes your issue

daniel-j-h, 2017-01-25:
The new units just arrived and I could reproduce the issue there!
For the record the old unit's serial no. was in the two digit range, this is in the three digit range.

I created a pull request fixing this issue and could confirm it working now on the recent units I have here.

dcyoung, 2017-01-25:
Looks good, I was using the same internal fix myself. This resolves issues with subsequent runs. This issue can be effectively closed when the PR is merged.

The `example-c++` and `example-viewer` are still throwing an occasional (read ~10% of the time) error...

> unable to receive stop scanning command response

However, this is being thrown on program exit, not startup. From some early debugging, it looks like this is because the sleep duration in the `sweep_device_stop_scanning` method is occasionally too short. Increasing it a bit resolves the issue, but is NOT the ideal fix. This is related to a different issue, continue it here #27

commit 860e4170 (2017-01-25): Stops scanning and flushes serial device on destruction, resolves #23
commit f2cb07e2 (2017-01-26): Stops scanning and flushes serial device on destruction, resolves #23
closed 2017-01-26 by daniel-j-h