scanse/sweep-sdk #26: Add full semantic version scheme

dcyoung, 2017-01-24:
Currently, we only use major and minor version scheme. Any reason we shouldn't add support for full semantic version scheme... ie: Major.Minor.Patch ?

daniel-j-h, 2017-01-24:
Do you mean the soname versioning [here](https://github.com/scanse/sweep-sdk/blob/b4e6e8e4c4d7f64af2efd5baf3bf62ae634ceb6f/libsweep/config.mk#L9)? It's for specifying ABI compatibility and letting users link against one library and later simply change in a ABI compatible more recent library.

Here is a nice intro:

https://www.bottomupcs.com/libraries_and_the_linker.xhtml

The SDK itself or libsweep could be versioned via semantic versioning at the same time, of course.

daniel-j-h, 2017-02-18:
I hope the switch to CMake makes it more clear now: we [do have semantic versioning in place](https://github.com/scanse/sweep-sdk/blob/334f74aab5eb716c17c466cb2718920ed1d88f09/libsweep/CMakeLists.txt#L7-L9). We just have to make sure to move and upgrade `libsweep` and bindings in lock step.

[Here's](https://github.com/scanse/sweep-sdk/blob/334f74aab5eb716c17c466cb2718920ed1d88f09/libsweep/CMakeLists.txt#L62-L63) how we set version and more importantly soname for ABI compatibility.

I'm closing this as resolved. Feel free to re-open if needs be.

closed 2017-02-18 by daniel-j-h