scanse/sweep-sdk #35: "Error: unable to receive stop scanning command response" in example.cc

matpalm, 2017-01-27:
i'm running `libsweep/examples/example.cc` to get some calibration data and often find that after a quick stop/start i get `Error: unable to receive stop scanning command response` or `Error: unable to receive start scanning command response` 3 or 4 times before getting connectivity again. Unplugging the device and replugging it back in always seems to clear the problem (i.e. I don't get this error the first time I run the example). 

Is this expected? Is there additionally debugging I can provide?

dcyoung, 2017-01-28:
Could well be related to #27 

A temporary fix is described there (increasing the sleep duration before sending the second stop command). 

This will soon be replaced by a proper parse/check for the `DX receipt`. However, the implementation for that will have to come after we introduce a worker thread which accumulates the scans. This is in the works and should be accomplished soon. For now try increasing the sleep duration and see if that resolves your issue.

Try bumping up the sleep duration in this line:
https://github.com/scanse/sweep-sdk/blob/f2cb07e24eba08e651f4ca9c42763f1bbdaf3587/libsweep/sweep.c#L151

Maybe try doubling or increasing by an order of magnitude. Value is in microseconds, so its still a tiny pause for something like stopping the device.

daniel-j-h, 2017-02-05:
@matpalm can you change the `sleep` call from `0.005s` to x 2 the amount (and if it still does not work x 10) and report back please. I would love to get a sense of how long we have to wait and if this is a feasible hack^W workaround instead of implementing data packet decision logic ourselves.

matpalm, 2017-02-06:
sure. won't be able to until Friday though sorry, bit busy

daniel-j-h, 2017-02-18:
Any updates here? We're waiting for [20ms](https://github.com/scanse/sweep-sdk/blob/334f74aab5eb716c17c466cb2718920ed1d88f09/libsweep/src/sweep.cc#L140-L141) now to give the hardware a chance to stop sending packets.

matpalm, 2017-03-17:
sorry for the super late response :( 

grabbed latest sweep-sdk and not seeing this anymore from `while true; do ./example-c /dev/ttyUSB0; done`

fix looks good, thanks!!

closed 2017-03-17 by matpalm