scanse/sweep-sdk #37: Avoid using 'sizeof()' when specifying number of bytes to read

dcyoung, 2017-01-30:
When calling `sweep_serial_device_read` and potentially elsewhere, sizeof() is used to specify a number of bytes. 

Ex: 
https://github.com/scanse/sweep-sdk/blob/f2cb07e24eba08e651f4ca9c42763f1bbdaf3587/libsweep/protocol.c#L188

Something like `sizeof(sweep_protocol_response_scan_packet_s)` ends up being platform dependent, returning 7 bytes on linux and 8 bytes on windows. The input (`len`) to `sweep_serial_device_read` really shouldn't depend on the size of `sweep_protocol_response_scan_packet_s` which differs by OS. It should be based on the communication protocol spec and we should probably define this explicitly. Perhaps we should create a mapping of receipt types to expected length (num bytes) and pull `len` from there.

dcyoung, 2017-01-30:
Found the issue, described here:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=52991

Temporary workaround is to add compile flag `-mno-ms-bitfields`. 

I still think we might want to avoid using packed structs, and avoid basing the byte quantities off of packed structs. A more explicit "expected length" value for each type of command would be easier for new users to understand.

daniel-j-h, 2017-01-31:
Interesting issue. This is another use-case for switching to the C++11 impl. where we could

```c++
static_assert(sizeof(T) == N, "type does not conform to spec");
```

For now I think passing the flag you found is the right way to go. You have to detect and dispatch in [config.mk](https://github.com/scanse/sweep-sdk/blob/master/libsweep/config.mk) conditionally setting the flag.

daniel-j-h, 2017-02-05:
I pushed a changeset today for `static_assert`-ing all struct sizes: https://github.com/scanse/sweep-sdk/pull/33#commits-pushed-afd4421.

Once we merged the `c++impl` branch I would love to keep them in place. If MinGW is broken but the flag you mentioned fixes it, I'd just add the flag to the platform specific compilation flags.

dcyoung, 2017-02-06:
@daniel-j-h 
Thanks Daniel, I'll try to test this out tomorrow.

dcyoung, 2017-02-07:
The compiler flag was working fine when GCC was the only intended compiler. However that might change soon.

I realized today that MinGW has issues with c++11 std threads. There are workarounds, such as forcing MinGW to use a posix thread API model to interpret the c++11 thread calls, which it then converts to win32 thread calls via an abstraction that provides a pthread API. But this gets messy, and requires a fair amount of documentation. I'd like to avoid it.

Unlike GCC, the Visual C++ compiler is capable of directly handling c++11 thread calls on windows, without any intermediate posix interpretation. This is good news, as I've been meaning to add Visual Studio support for a while now anyways. I'd really like to present VS support as the main form of windows compatibility.  

Now, how does this relate to struct packing....? I started work on this today, but hit a snag related to packing structs. It seems that there is no direct Visual C++ equivalent of GCC's `__attribute__((__packed__))`. If there is one that will work just fine, I'm happy to use it. [This stackoverflow](http://stackoverflow.com/questions/1537964/visual-c-equivalent-of-gccs-attribute-packed) discusses some alternatives, but these seem like tedious workarounds for a problem we could easily avoid.

@daniel-j-h  What is the main motivation for packing the structs, and using the packed size to indicate the number of bytes in a message? Given that packing is rather platform dependent, it seems odd that we are exerting effort to use the size as an indication of message length. The implementation is all hidden behind our API, so why don't we simply add a value for message length and not worry about packing the structs?

dcyoung, 2017-02-07:
My hacky workaround idea was something like this:
```c
#if __GNUC__ >= 4
#define SWEEP_API __attribute__((visibility("default")))
#define SWEEP_PACKED2( class_to_pack ) class_to_pack __attribute__((__packed__))
#define SWEEP_PACKED __attribute__((packed))
#else
#if defined _WIN32 || defined __CYGWIN__
#define SWEEP_API __declspec(dllexport)
#define SWEEP_PACKED(class_to_pack) __pragma(pack(push, 1)) class_to_pack __pragma(pack(pop))
#else
#error "Only Clang, GCC and Visual supported at the moment, please open a ticket"
#define SWEEP_API
#define SWEEP_PACKED
#endif
#endif
```
which worked fine when using the Visual Compiler, but gcc spit out the following warning...

> warning: '__packed__' attribute ignored [-Wattributes]

Again, this all seems like unnecessary added complexity. Whats the justification?

daniel-j-h, 2017-02-07:
The idea behind packing structs is making them 1:1 memory wise with what we send over the wire.

The following has to be checked without packing:

```c++
struct S { uint8_t x,y; };

static_assert(std::is_trivial<S>(), "");                        // 1
static_assert(sizeof(S) == sizeof(S::x) + sizeof(S::y), "");    // 2
static_assert(sizeof(S) == 2, "");                              // 3
static_assert(offsetof(S,x) == 0, "");                          // 4
static_assert(offsetof(S,y) == 1, ""); }                        // 5
```
- 1/ checks if it's trivially copyable, basically if we can memcpy an instance of it
- 2/ total size is defined as sum of parts
- 3/ total size is statically known
- 4/ x is at the front
- 5/ y comes after x

3/, 4/ and 5/ ensure packing with no padding in between.

I'm not sure how to do this on Windows - your approach looks like a good (but ugly :P) solution.

All we need is the `static_assert` for the struct's in memory representation to hold. How else can we guarantee for reading / writing the correct bytes?

dcyoung, 2017-02-07:
I see what you mean there. I was thinking more that we'd read to and write from specific entries in the command structs rather than assume they're packed. But as I think more about it, this is not ideal for performance... which might become more important for future sensor with drastically different scan rates.

Any ideas for how to modify or use a wrapper like
```c
#define SWEEP_PACKED( class_to_pack ) class_to_pack __attribute__((__packed__))
```
such that the attribute won't be ignored with gcc? 

If not, should we just implement an extremely ugly #if/#else  in `protocol.h` that defines structs in different ways?

MikeGitb, 2017-02-07:
@daniel-j-h: Realistically speaking: If all your members are uint8_t and you don't target some really unusual architecture all of this will always be true without packing. In particular //1 is guaranteed by the standard, and the order is too, so effectively you only need test 3.

Also if you go crossplatform: The meaning of packed is not standardized accross different toolchains.

dcyoung, 2017-02-08:
So this works with MinGW using gcc, and through Visual Studio using Visual compiler. It's not very elegant, but are there any objections?

In `sweep.h`
```c
#if __GNUC__ >= 4
#define SWEEP_API __attribute__((visibility("default")))
#define SWEEP_PACKED __attribute__((packed))
#else
#if defined _WIN32 || defined __CYGWIN__
#define SWEEP_API __declspec(dllexport)
#define SWEEP_PACKED(class_to_pack) __pragma(pack(push, 1)) class_to_pack __pragma(pack(pop))
#else
#error "Only Clang, GCC and Visual supported at the moment, please open a ticket"
#define SWEEP_API
#define SWEEP_PACKED
#endif
#endif
```
and `protocol.h`
```c
// Packets for communication

#if __GNUC__ >= 4
typedef struct {
  uint8_t cmdByte1;
  uint8_t cmdByte2;
  uint8_t cmdParamTerm;
} SWEEP_PACKED sweep_protocol_cmd_packet_s;
...
#else
#if defined _WIN32 || defined __CYGWIN__
SWEEP_PACKED(typedef struct {
  uint8_t cmdByte1;
  uint8_t cmdByte2;
  uint8_t cmdParamTerm;
} sweep_protocol_cmd_packet_s;)
...
#else
#error "Only Clang, GCC and Visual supported at the moment, please open a ticket"
#define SWEEP_API
#define SWEEP_PACKED
#endif
#endif
```

MikeGitb, 2017-02-08:
I'd really rather see all those `packed` modifiers removed completely.

daniel-j-h, 2017-02-09:
@MikeGitb what is your proposal then? Remove all packed attributes and have the structs be `n` bytes in memory. Then read `m` bytes of packets from the device, and fill in struct members manually?

MikeGitb, 2017-02-09:
@daniel-j-h: Maybe I'm missunderstanding something, but what I'm proposing is just remove them and not changing anything else. AFAIK they don't have any effect for those structs.

MikeGitb, 2017-02-10:
@daniel-j-h: Sorry, my mistake. I erroneously assumed, that all structs would have only `uint8_t` members, in which case their sizes wouldn't change. As it stands there is a single structure that is affected by the  packing: Namely `response_scan_packet_s` which has two `uint16_t` fields ( `angle` & `distantce`).

One possibility to solve that problem without packing would be to define those fields as arrays of `uint8_t` and then use a helper function to access them:	

    typedef struct {
		uint8_t sync_error;
		uint8_t angle[2]; // see u16_to_f32
		uint8_t distance[2];
		uint8_t signal_strength;
		uint8_t checksum;
	} response_scan_packet_s;

	uint16_t combined(const uint8_t(&d)[2]) {
		return d[0] | d[1] << 8;
	}

Admittedly that is not really better than some macro to unify the packing syntax.

MikeGitb, 2017-02-12:
OK, so I just found out, that gcc and clang support 

    #pragma pack(push, 1)
    ...
    #pragma pack(pop)

as well, so why not stick with that syntax?

daniel-j-h, 2017-02-13:
If it works for Windows (VS + MinGW), too, I'd go for adding it to the top and end of our structs files :+1:

MikeGitb, 2017-02-13:
I don't know about MinGW, but it is the *native* syntax for MSVC and definetively works there. Btw. [godbolt's compiler explorer](https://godbolt.org/) now also supports VS. It's a great tool to check which compilers supports a certain syntax/feature.

dcyoung, 2017-02-13:
I switched out the attribute style packing in favor of push/pop. Pushed in the latest win commit, as its only necessary once the windows PR is merged: 9e9319f

I tested this on both MinGW-w64and MSVC, and it builds just fine on the Travis linux machines. Does this solve our problem?

daniel-j-h, 2017-02-14:
Perfect, then let's push the state at the file's top and pop it again at the bottom!

commit 4c12c07c (2017-02-16): Makes Struct Packing Platform Independent, resolves #37.
closed 2017-02-16 by daniel-j-h