scanse/sweep-sdk #50: Switch libsweep from Makefile to CMake

daniel-j-h, 2017-02-13:
We're currently using a `Makefile` + `config.mk` for the build system. Now with the implementation switching to C++11 and @dcyoung polishing up the Windows branch we should re-visit this decision.

- [Makefile](https://github.com/scanse/sweep-sdk/blob/6c3ae36af4a1ae6cb5dee1444850728a96dc4497/libsweep/Makefile)
- [config.mk](https://github.com/scanse/sweep-sdk/blob/6c3ae36af4a1ae6cb5dee1444850728a96dc4497/libsweep/config.mk)

The examples are using CMake, too, and we provide a [`FindSweep.cmake`](https://github.com/scanse/sweep-sdk/blob/master/libsweep/cmake/FindSweep.cmake) CMake module already.

[Here](https://github.com/scanse/sweep-sdk/blob/6c3ae36af4a1ae6cb5dee1444850728a96dc4497/libsweep/examples/CMakeLists.txt#L7-L9) is how users can then include `libsweep` into their CMake based build system.
Related issue: [install FindSweep.cmake on the user's system](https://github.com/scanse/sweep-sdk/issues/17).

Should probably land before merging the Windows support branch to eliminate the need for Visual Studio specific project files, if we decide on going this route.

Extra requirements for a CMake based build system:
- Require CMake 2.8.12, so we can build without having to upgrade CMake going back to [Trusty](http://packages.ubuntu.com/trusty/cmake)
- Add SONAME (or equivalent) to shared library in a portable way
- Provide install target for `libsweep.so`+`sweep/sweep.h`+`sweep/sweep.hpp`

dcyoung, 2017-02-14:
Did some early testing and seems like cmake produces VS solution just fine. Some things to note...

- During this conversion from makefile to cmake, we might want to decide on a src tree structure. 
- Can we make the examples agnostic to the installation style that differs by platform.
- Instead of defining version details etc in `sweep.h`, we could have `sweep.h` include a config header file that is populated with version details by cmake.

daniel-j-h, 2017-02-14:
Usually the source tree looks similar to

    src <- impl. in C++11
    include <- headers for impl.
    include/sweep <- public sweep.h and sweep.hpp

The install target then simply copies over the library and include/sweep directory.

I agree with you, we should let CMake's configure_file read config.h.in and generate config.h.

commit 041ac458 (2017-02-14): Switches libsweep from Makefile to CMake, resolves #50
commit 09dbfc2f (2017-02-14): Switches libsweep from Makefile to CMake, resolves #50
commit bc995b77 (2017-02-15): Switches libsweep from Makefile to CMake, resolves #50
commit 37da039e (2017-02-16): Switches libsweep from Makefile to CMake, resolves #50
closed 2017-02-16 by daniel-j-h