scanse/sweep-sdk #56: Broken C++ example

dcyoung, 2017-02-16:
It looks like the sample rate commit broke the c++ example by placing the calls to get_motor_speed and get_sample_rate AFTER `start_scanning`...

https://github.com/scanse/sweep-sdk/blob/4c12c07c16c6fc82f49fed78929119f1db1e1b07/libsweep/examples/example.cc#L11

This probably works with the dummy library because it sends back information regardless of whether or not its scanning. However, the actual sweep device cannot receive or respond to commands other than `DX\n` while it is scanning. The calls to `get_motor_speed` and `get_sample_rate` need to be placed before `start_scanning`. Additionally, in those functions we should probably check if the device is scanning and simply error or report it somehow before trying to message a device... which will only cause more errors.

MikeGitb, 2017-02-16:
I don't have a strong feeling either way, but how about starting to scan automatically?

daniel-j-h, 2017-02-16:
Hm. I followed the initial design and giving that the library is quite low-level having start and stop functions on the device seemed like a good idea. I can see how the constructor should start and the destructor should stop, making use of the device's lifetime.

On February 16, 2017 7:12:37 PM GMT+01:00, MikeGitb <notifications@github.com> wrote:
>I don't have a strong feeling either way, but how about starting to
>scan automatically?

MikeGitb, 2017-02-16:
I meant Autostartimg when getScan is called.

dcyoung, 2017-02-16:
Hmmm. I think there is some confusion here. The device should definitely be created without auto starting the scanning process. ie: we should be able to explicitly start and stop the scanning as we currently do. 

The only issue is that the C++ example sends two other commands (get_motor_speed, and get_sample_rate) AFTER the scanning is started. When designing a user application, one should never send a command other than STOP to the device while it is scanning. However, since it is possible to explicitly call the functions when the device is scanning, we are seeing errors (likely originating in the read function timing out because the device never responded). 

We should prevent this case by never allowing the command to be sent if the device is scanning. Ie: in `get_motor_speed` and all other functions that send commands (other than stop_scanning), we should check that the device is NOT scanning before sending a command.

dcyoung, 2017-02-16:
I went ahead and pushed the example-c++ fix... but I'm leaving this open and changing the title for discussion regarding the addition of checks for `is_scanning` in all functions that send commands.

daniel-j-h, 2017-02-16:
Agree. We have to assert the device not scanning for those functions as a precondition. For dummy and real library. And make it clear in docs.

On February 16, 2017 8:12:38 PM GMT+01:00, dcyoung <notifications@github.com> wrote:
>I went ahead and pushed the example-c++ fix... but I'm leaving this
>open and changing the title for discussion regarding the addition of
>checks for `is_scanning` in all functions that send commands.

dcyoung, 2017-02-16:
Sorry, accidentally closed and reopened the issue. 

@daniel-j-h  PR is up at #57, go ahead and merge if all looks good.

closed 2017-02-16 by daniel-j-h

MikeGitb, 2017-02-16:
If the device doesn'T accept any commands while in scanning mode then what happens, if the program gets terminated or simply doesn't stop the device at the end and on the next program start, the device is already scanning? Maybe the API should always send a stop command first. I think an assert is counterproductive here.

daniel-j-h, 2017-02-16:
We already do that see [here](https://github.com/scanse/sweep-sdk/blob/334f74aab5eb716c17c466cb2718920ed1d88f09/libsweep/src/sweep.cc#L72).

That's the reason the `stop_scanning` does not assert for `is_scanning`.

It may seem like a workaround but with a stateful hardware device there is only so much we can do.

dcyoung, 2017-02-16:
I believe the current firmware also supports the reset command while the device is scanning. We are debating whether or not to support this, or assume that anytime a user will reset a device, they can explicitly check and call stop scanning.

PR #57 (2017-02-16): Check for scanning device before sending cmd
This PR resolves #56