scanse/sweep-sdk #61: Project structure

MikeGitb, 2017-02-19:
This might be specific to my particular needs but I'd find it more intuitive and useful if things like the examples and docs would be put into a separate folder under the root directory, keeping the library directories clean:

````
-sweep-sdk
     |-libsweep
     |    |-include
     |    \-src
     |    
     |-sweepjs
     |-sweeppy
     |-docs
     |    \-serial_protocol_spec.md
     |
     |-examples
     |   |-c
     |   |-cpp
     |   |-js
     |   \-python
     |
     \-CMakeLists.txt  <-- master CMakeFile
````

Also, I mentioned this in a PR but I'd also prefer it, if the examples could be compiled without having to install the liibsweep library first. 

E.g. on windows it is not clear cut, where such a library should be installed and in partucular if one just wants to have a quick look, I think it is preferable not to require any changes to the system. Pretty much the same might hold true for many embedded systems for which you need to use a crosscompiler.

kent-williams, 2017-02-19:
+1 for moving the docs folder to top dir!

daniel-j-h, 2017-02-19:
What is the master `CMakeLists.txt` supposed to do?
Install libsweep and all bindings?
Also build and run examples?

Python and Js bindings are getting built and installed via setuptools and node-gyp, respectively.
Do you want to shell out via CMake?

The reason for having a docs directory under libsweep is it contains low-level docs _only needed for implementing libsweep_. Binding users (and even library _users_) should not have to care about these internals - that's the reason we abstract over them in the first place.

kent-williams, 2017-02-19:
Valid point on the docs location. I'm thinking from the perspective of someone just wanting to find the protocol, but it doesn't really make sense for it to be easily found here anyways.

MikeGitb, 2017-02-20:
@daniel-j-h: The master CMake file should ensure that libsweep is build before any example or binding is and in general provide all make targets (although functionaly they would be implemented in the subdirectories). It imho should not install the library unless explicitly asked to by the user.

From what I'm used to by other libraries I'd like to be able to

   clone ...
   cd sweep-sdk
   mkdir build; cd build
   cmake ..
   make all
   ./bin/example-c++
   ./bin/example-python

MikeGitb, 2017-02-20:
Concerning the installation: I'll most likely use this library in a project running on 3 different machines:

- A windows machine
- A desktop linux machine without root access
- An embedded linux machine running on a xilinx zynq (arm core) for which all programs are compiled using a crosscompiler (from the desktop linux). 

For none of those platforms can or want I install this library, but intend to just copy libsweep in a subfolder and build it together with the rest of the program. Now, of course I can make the necessary adjustments myself, but I find it hard to believe that I'm the only one that has such a use case.

MikeGitb, 2017-02-20:
Regarding the docs: I see your point. This is probably just a personal habit of mine, but I usually try to keep the source tree clean of any documents (be it developer or user documentation). It is probably a question of whether one sees the whole repository as a single project or as three individual ones. That particular document might btw. also be interesting for people who have a raw data dump or want to talk to it via the console (for whatever reason)

daniel-j-h, 2017-02-21:
For the C and C++ examples we can simple use CMake's `add_subdirectory()` and have recursive CMakeLists.txt's, that would be fine. I'm more concerned about the Node and Python bindings.

How do we want to build and run them without installing `libsweep` when their build system and runtime searches for the library and include paths globally? For the Node bindings we would have to make the search paths configurable from the command line and then pass the local directories. Same for the Python example. In both cases users have to adapt `LD_LIBRARY_PATH` then before running the examples.

Btw if your use-case consists of cross-compiling for a xilinx zynq you probably know what to do already :)

MikeGitb, 2017-02-21:
@daniel-j-h: Unfortunately I don't have experience with Node and Python dindings and wasn't aware of that limitation. I certainly don't want to make things easier for the user.

You are right that I've probably a very particular view on this that might not match the primary target of this library. But - at least for c/c++ programs - it has been my experience that being able to compile everything together from the ground up makes life much easier when portability is a concern (Although that is probably much more of a concern on windows and embedded devices than a standard linux).

closed 2017-02-21 by MikeGitb