scanse/sweep-sdk #72: Creating sweep object hangs

damyvv, 2017-04-26:
When we connect the sweep to our usb port it opens on /dev/ttyUSB0. The visualizer program works flawless with the sweep. So we took the example.cc file and succesfully compiled the project. But when we run the example with "./Lidar /dev/ttyUSB0" it prints "Constructing sweep device..." and then just hangs. It doesn't throw any error, but it's not proceeding. Looking at your code I would expect it to output its motor speed, but it doesn't do that. So it doesn't get past following line:
[https://github.com/scanse/sweep-sdk/blob/master/libsweep/examples/example.cc#L16](url)

If we disconnect the sweep it throws the "Error: opening serial port failed", so we know "/dev/ttyUSB0" is the right port.

Thanks in advance.

dcyoung, 2017-04-26:
Hi there damyvv, could you provide a little more info about your setup?

- What operating system are you using?
- What firmware is installed on your sweep device (you can check this using the sweep visualizer)?
- Is `example.cc` the only example causing issues? Does `example.c` work as expected?
- Is the issue with `example.cc` intermittent or does it hang every time you try it?
- Could you clarify how you are calling the example. I'm a little unclear where the `./Lidar` is coming from?

damyvv, 2017-04-27:
Hi dcyoung,

Thanks for your quick response. I can't access the sweep until monday, but I will answer your questions as soon as possible.

For now I can only provide this information:
- We are using ubuntu version 14.04 (also tried it with ubuntu 16.04 on a different computer, but it had the same result)
- The example.cc hangs every time. Tried it about ten times.
- We compiled the example.cc to an executable and we called it 'Lidar'. So that is just the compiled example.cc file

daniel-j-h, 2017-04-30:
Can you build both `libsweep` as well as your example with debug information, i.e.

    cmake .. -DCMAKE_BUILD_TYPE=Debug

and then run under `gdb` and give us a backtrace (`bt`) of where exactly it hangs? You can also attach to your running and hanging example program if that makes it easier for you.

If you compiled the [`example.cc`](https://github.com/scanse/sweep-sdk/blob/404a2d65c59e42b2b51c1a4817d8b881c864cee4/libsweep/examples/example.cc#L16) as is it most likely hangs either during serial device construction or sending the initial stop command.

Serial port issues could be related to your user not being allowed to access the serial devices, as in you're not in the `dialout` group:

    sudo usermod -aG dialout $(whoami)

But that's jsut a wild guess at the moment - need more information.

damyvv, 2017-05-01:
Updating the firmware to version 1.3 solved the problem. Thanks for your support.

damyvv, 2017-05-08:
Hi,

I'm not sure you have seen my edited post. Did I provide you with enough information?

Thanks.

daniel-j-h, 2017-05-08:
Thanks, the backtrace is surprisingly readable and helpful!

What happens is this:
- We create the device object
- We open the serial port
- We send an initial stop command [here](https://github.com/scanse/sweep-sdk/blob/master/libsweep/src/sweep.cc#L71-L72) to stop the device just in case it's already scanning
- We never get a response and hang in the wait loop [here](https://github.com/scanse/sweep-sdk/blob/9b07012851da185beccd94f6af03267d5b1572c8/libsweep/src/unix/serial.cc#L330)

@dcyoung @kent-williams any idea why the firmware might not send a reply to a `DX` command? Do we need to flush the serial port just in case before sending the stop command?

dcyoung, 2017-05-08:
The only reason it wouldn't respond to a DX command is if the command was sent while the device is still booting up (green led on the face of the device is flashing). Be sure the led has turned blue before creating a device. Other than that, we've never seen it fail to respond to a DX. 

Is the device working fine with the visualizer?

dcyoung, 2017-05-23:
@damyvv  Were you able to resolve this?

damyvv, 2017-05-23:
Well... No unfortunately not. But the thread Daniel provided did sort of help. 

We weren't using the provided FTDI. But we weren't using that for a reason, I will get back to that later. I tested again on a different machine with the original FTDI. And it works perfectly with the SDK. Now we were using a different USB-serial converter the CP210x. This chip wasn't working with the SDK, but the weird thing is it works fine with the visualizer!

The reason we didn't use the original FTDI is because we are using a ARM Linux machine and it was missing the drivers. We tried downloading the VCP drivers from the FTDI website, but we weren't able to compile the drivers for our platform. The CP210x did have drivers for out platform, so that is why we chose to use that instead.

Vrong, 2017-07-17:
Hey I have the same issue, it sends the stop scanning command and then nothing. It blocks on the "select" statement when reading the serial and it's like it never get something to read. 
The thing is that if I spy with cutecom I can see the DXOOP responses that the API doesn't see. 
So I guess there is definitely an issue with the API.
I'm using the provided FTDI on a Raspberry Pi 3.

dcyoung, 2017-07-17:
I can confirm this behavior as well running on a Raspberry Pi 3, although it appears to be intermittent. As Vrong mentions, the hang occurs on the select call [here](https://github.com/scanse/sweep-sdk/blob/0a326fa2f36bc1ab9b1f878deee85714bec62807/libsweep/src/unix/serial.cc#L40)

I'm not very familiar with the unix serial implementation.  I'm I haven't been able to find a good lead on why the incoming data can't be seen/read. @daniel-j-h any ideas here? Perhaps its related to issues discussed in #73.

Is there a better or more robust method for performing the unix reads? At the very least we should include a timeout to avoid the hang (see [select](http://pubs.opengroup.org/onlinepubs/009696899/basedefs/sys/select.h.html)), yeah? This might be relevant to discussion in #109.

Vrong, 2017-07-17:
Ok I fixed it, you can use changes from my pull request above or my repository directly : https://github.com/Vrong/sweep-sdk 

Have fun !

closed 2017-07-18 by dcyoung

dcyoung, 2017-07-19:
Fix is incorporated in the most recent release `v1.2.2`.  Closing issue.

daniel-j-h, 2017-07-22:
Sorry for being late here - agree with you @dcyoung that timeouts would be the way to go here. I think we initially had timeouts but then removed them because we were under the impression these issues would never occur with a working protocol implementation. A hanging API function is terrible for users integrating libsweep.

vitorboschi, 2017-08-07:
Even with v1.2.2 I have the exact same issue described by OP.
I've tracked the problem down to the code wrongly setting character size and after fixing that (PR #119) everything seems to work for me.