scanse/sweep-sdk #73: DX receipt left in buffer

dcyoung, 2017-04-27:
I've discovered an issue but not entirely sure what is causing it.

- OS: linux
- Platform: so far, I can only reproduce this behavior on the raspberryPi
- Library: libsweep (and by extension sweepy + sweepjs as well)
- Repeatable: around 25% of the time

Code to reproduce:
```c++
sweep::sweep device{argv[1]};
device.set_sample_rate(500); // errors here, description below
```

The expected behavior should be as follows:

- During device creation, `sweep_device_stop_scanning()` is called internally... which performs the following:
  - sends `DX\n` command to stop any potential active data stream
  - waits 35ms to allow any incoming data blocks, and the DX response (`DX00P\n`) to come in
  - flushes buffer (including the first DX response)
  - sends another `DX\n` command
  - reads the DX response (`DX00P\n`) that should be responding to the 2nd command
- Set sample rate
  - `sweep_device_set_sample_rate()` sends `LR01\n` command to set the sample rate and then reads LR response (`LR01\n00P\n`)

The issue is a checksum error when attempting to read the LR response. Instead of `LR01\n00P\n`, the response that is read from the serial buffer is `DX00P\nLR0`. It appears that a `DX - Stop Scanning` response (`DX00P\n`)  is being left in the serial buffer, and is therefore being interpreted as the beginning of the LR response. 

Using a protocol analyzer I have confirmed that the timing of the commands and responses, as well as their byte content, is as expected:
Here is a zoomed out timeline:
- device is constructed (sends 2 DX commands 35 ms apart, both receive valid responses)
- Set sample rate (with valid response)
- device is destructed (sends 2 DX commands 35 ms apart, both receive valid responses)

[image]

From this I can tell that the communication over the wire is fine. No extra "DX" response (`DX00P\n`) is coming in, and no response is delayed. The bytes from the "LR" response (`LR01\n00P\n`) are valid as well. 

It seems then that somehow the buffer isn't being flushed properly. @daniel-j-h Any idea what could be causing this?

dcyoung, 2017-04-27:
I took a peek at the contents of the buffer at various stages of the `sweep_stop_scanning()` method using `ioctl(serial->fd, FIONREAD, &bytes);`. 

I inserted this check immediately after the flush and then again after reading the second response. ie:
```
void sweep_device_stop_scanning() {
  // send first stop command
  // Wait 35ms

  // CHECK #1: check num bytes in buffer

  // Flush
  
  // CHECK #2: check num bytes in buffer

  // Write another stop cmd so we can read a response
  // read the response

  // CHECK #3: check num bytes in buffer
}
```

What we expect to see is:
`CHECK #1` should show a minimum of 6 bytes (any trailing data stream + `DX00P\n` response to first stop command) 
`CHECK #2` should be an empty buffer
`CHECK #3` should be an empty buffer

However, I'm seeing the following:
`CHECK #1` reports an empty buffer (0 bytes in buffer)
`CHECK #2` reports an empty buffer (0 bytes in buffer)
`CHECK #3` reports 6 bytes in the buffer (ie: a match for `DX00P\n`)

To me this suggests the the flush is not the issue... instead it appears that the incoming bytes are not being buffered right away. The protocol analyzer showed that the `DX00P\n` responses are sent over the wire with the correct timing, but perhaps there is some unknown delay before they are being placed in the buffer? Could this be because during device construction, we call `sweep_device_stop_scanning()` so soon after creating the serial device? This seems unlikely.

dcyoung, 2017-04-28:
I have found a pseudo solution. Currently we wait for the device to respond to the stop command before flushing. Although the actual respose is sent over the wire almost instantly, that does not guarantee that the host's serial setup has properly handled those incoming bytes. Apparently on a RaspberryPi, even the 35ms pause between the two stop commands was not enough at times. 

To fix this, I modified the `sweep_device_stop_scanning()` command to do the following:
```c++
void sweep_device_stop_scanning() {
  // send first stop command
  // Wait 35ms

  // INSERT: Read the potentially garbage response, and just discard it

  // Flush
  // Write another stop cmd so we can read a response
  // read the response
}
```

The possible conditions and the resulting behavior of the inserted code are as follows:
- Condition 1: data stream was active before sending the first stop command
  - Attempting to read the response will result in an error, because the leftover data stream bytes will pass a check for a valid response.
  - Destroy the error
  - Proceed on (flush buffer + send second DX and read valid response)
- Condition 2: data stream was not active, and RaspberryPi moved the `DX00P\n` response into the buffer during the 35ms pause
  -  The response will validate properly, without error 
  - Proceed on (flush buffer + send second DX and read valid response)... we do this because technically the random data blocks could masquerade as a valid stop response
- Condition 3: data stream was not active, and the buffer is empty at the end of the 35ms pause (ie: RaspberryPi did not yet move the `DX00P\n` response into the buffer)
  - The read will wait for the response to be moved into the buffer
  - The response will validate properly, without error
  - Proceed on (flush buffer + send second DX and read valid response)... we do this because technically the random data blocks could masquerade as a valid stop response

I'm not sure if its possible, but there might be a possible edge case where the data stream is active, so there are bytes in the buffer... but the RaspberryPi did not yet move the `DX00P\n` response into the buffer. In that case, the garbage read doesn't wait for the `DX00P\n` response, but instead just reads the existing garbage bytes. Then the flush that comes after the garbage read might not catch the response and the original issue persists.... but I haven't been able to produce this edge case. For now this seems like a better solution than what we have currently. I'll create a PR shortly.

daniel-j-h, 2017-04-30:
Hmm not sure I fully understand this - the solution seems to just read the data and discard it. Isn't this what our flush command should do?

I'm also happy to just increase our timeouts if it makes things easier - `35ms` seems arbitrary anyway. The fact that we have host-dependent timeouts all over the place is ugly but there is no way around it, is there?

dcyoung, 2017-05-01:
> the solution seems to just read the data and discard it. Isn't this what our flush command should do?

That is exactly what the solution is doing. The flush command "should" do this... but for some reason on the raspberry pi, the data isn't in the buffer for an odd amount of time. Attempting to read the response from the first `DX` basically waits for the data to arrive, whereas flush does not. 

> I'm also happy to just increase our timeouts

Increasing the timeouts isn't ideal here, as that still does not accomplish a proper wait for the data... it just assumes the data would be in the buffer by then. After adding this fix, the new read occasionally had to wait almost a second for the response to appear in the buffer... despite it being sent over the wire within milliseconds. I'm guessing this is some problem with the Pi's serial setup... but the added read properly waits for the data to be in the buffer, whereas increasing the sleep time does not. 

> 35ms seems arbitrary anyway

The 35 ms isn't arbitrary. It was added as a safe buffer to account for the ~25ms logging procedure that the device performs when stopping data acquisition. This may have changed in later firmwares, but during the logging process, the device won't be able to respond to the 2nd stop command. We add the 35ms pause in case the device was scanning and the first `DX` triggers this logging process. The pause ensures we don't send the second `DX` until the device is done logging session info. Additionally, the pause permits the accumulation of all leftover data stream + response to the first DX command, before flushing. 

> there is no way around it, is there?

We could parse all incoming data. Its more cumbersome, and requires more careful attention to the guarantees + nuances of the communication protocol, but it is completely possible. This is how the js-sdk for the visualizer operates.

dcyoung, 2017-05-11:
Directly relevant: https://github.com/raspberrypi/linux/issues/1709

It looks like a fix has been merged into the RPi firmware, only a few days ago. Hopefully that will make its way into the next raspbian release, which might solve the problem at its source.  In the meantime, PR #74 should provide a temporary solution.

dcyoung, 2017-05-25:
Resolved in 63d30091396826817cf63f9db9652882643a9b14

closed 2017-05-25 by dcyoung