scanse/sweep-sdk #79: Angle unit conversion

brusangel, 2017-05-03:
_from sweep datasheet:_
>**Angle measurement** Sweep uses an optical encoder to measure the angle of the rotating sensor head. The angle that is recorded for a range data point is the angle the sensor is at when the measurement is completed.The beginning of the scan, and zero degrees is located where the status LED projects out of the base of the sensor

could we get provided with a formula to convert from the encoder "counter-ticks" to degrees?

or at least the encoder model, so that we can see the manufacturer information (ticks per degree or so).

kind regards!

dcyoung, 2017-05-03:
Hi there brusangel,

The sweep does not use a standard "encoder". The plastics of the device itself provides the ticks. It's a bit more complex than just a simple formula for "counter-ticks" to degrees, as the sensor performs calibration routines to adapt to inconsistencies in the physical ticks and properly interpolate angles between ticks depending on various settings.

You should never have to concern yourself with the method of encoding. The sensor communicates sensor readings with an azimuth value in degrees. See the section titled `Data Block Structure` from the  [protocol docs](https://github.com/scanse/sweep-sdk/blob/master/libsweep/doc/serial_protocol_spec.md)

The `libsweep` library performs all the parsing and provides simple methods to access the value in `millidegrees` which permits the return as an integer rather than a float.
`int32_t sweep_scan_get_angle(sweep_scan_s scan, int32_t sample)`

Sweeppy and Sweepjs also return the value in millidegrees.

dcyoung, 2017-05-05:
Closing this now. Feel free to comment here if you'd like to reopen the issue.

closed 2017-05-05 by dcyoung

brusangel, 2017-05-08:
thank you very much for your answer, I'm working with Python (sweeppy), so far I have a very simple code to extract lidar information into an standard array:

```
def main():
    dev = "/dev/ttyUSB0"
    with Sweep(dev) as sweep:
        sweep.start_scanning()
        for scan in itertools.islice(sweep.get_scans(), 2):
            data = []
            for row in scan:
                for celd in row:
                    dLidar=[celd[0],celd[1],celd[2]]
                    data.append(dLidar)
            print data
        #sweep.stop_scanning()
main()
```

the above code yields an output in the following style:
`[[2562, 119, 183], [4937, 131, 199], [7500, 132, 199], [10062, 128, 199], [12625, 109, 183], [15000, 123, 199], [17562, 108, 191], ... , [359250, 108, 191]]`

the first field as you may recognize is the positioning data, followed by distance and rss.

I've searched for the wrapper for the C function you mentioned `(sweep_scan_get_angle)`. but couldn't find it.

which can be the appropriate way to use the function in python?

thanks!

dcyoung, 2017-05-08:
The sweep_scan_get_angle is from the libsweep library and is used internally by the sweeppy python bindings. You won't have to call it directly when you use sweeppy. The returned samples that you posted are a collection of sensor readings. Each is comprised of 3 values (angle, distance and signal strength). The angle is presented in millidegrees.  For example, the frst reading you posted [2562, 119, 183] is a sensor reading with an angle of 2562 millidegrees (2.562 degrees), a distance of 119 cm and a signal strength of 183 (measured on a scale from 0 to 255). You shouldn't have to do any conversion work besides converting from millidegrees to degrees

daniel-j-h, 2017-05-08:
@brusangel [here](https://github.com/scanse/sweep-sdk/tree/master/sweeppy#interface) is the lib's interface. All the response object's are [`namedtuples`](https://docs.python.org/3/library/collections.html#collections.namedtuple) so I recommend using `sample.angle` instead of `sample[0]` etc.

If you are looking for a transformation to Cartesian coordinates, the real time viewer does something similar to:

```python
for sample in scan.samples:
    distance = min(sample.distance, k_max_laser_distance_m)
    angle = math.radians(sample.angle / 1000.)

    x = math.cos(angle) * distance
    y = math.sin(angle) * distance

    x = x + k_max_laser_distance_m
    y = y + k_max_laser_distance_m
```

Maybe we should just expose these helpers to users => https://github.com/scanse/sweep-sdk/issues/15

brusangel, 2017-05-09:
@dcyoung , @daniel-j-h guys! thanks for your help totally get it!

really appreciate this!

daniel-j-h, 2017-05-09:
This is used in the real time viewer to zoom into a couple of meters - you don't want to show the full 40 m range to the user, when the typical use-case is scanning a room.