scanse/sweep-sdk #85: Sweep locking up using C SDK

justin-p-m, 2017-05-08:
While using the C code SDK, My code uses the following libsweep.lib functions in this specific order:
sweep_device_construct(port, 115200, &error)
sweep_device_set_motor_speed(sweep, hz, &error)
sweep_device_set_sample_rate(sweep, sampleRate, &error)
sweep_device_get_motor_ready(sweep, &error)
int32_t speed = sweep_device_get_motor_speed(sweep, &error);
int32_t rate = sweep_device_get_sample_rate(sweep, &error);
sweep_device_start_scanning(sweep, &error);
check(error);
scan = sweep_device_get_scan(sweep, &error);
This is a similar setup to what is seen in example.c.
There are two issues that I run into. 
The first, when I use the settings of hz = 5 and sampleRate =1000, the scanse gets locked up within sweep_device_get_scan(). Following the callstack, I noticed that within scanse.cc -> class ScanQueue is where the lock up occurs while "the_queue" is empty. Is there a way to reset the scanse after a certain amount of time of it being locked up? It seem that the scanse will wait indefinitely, and at some point I would like the scanse to reset. 
The second issue occurs in the same place, however, when I use the setting hz = 5 and sampleRate = 500, the scans works as expected until a rotational acceleration is applied to the sweep. The motor monetarily stops, and seems to reset. However, the code gets locked up in the same place as described above. There are also cases where the sweep is running normally, and without any external disturbance, the code gets locked up in the same place.  I would like for it to continue scanning similar to what the visualizer does. When I perform the same rotational acceleration to the sweep while it is running in the visualizer, the motor resets and the visualizer continues to receive data after a momentary pause. Any advice what to do in order to achieve that using the SDK? 

I am using the version 1.3 of the sweep firmware, and the lastest version of libsweep.lib as of 5/8.
I am running Windows 7 x64

daniel-j-h, 2017-05-08:
> The first, when I use the settings of hz = 5 and sampleRate =1000, the scanse gets locked up within sweep_device_get_scan()

This should never happen and is definitely a bug.

Can you build both `libsweep` as well as your example with debug information, i.e.

    cmake .. -DCMAKE_BUILD_TYPE=Debug

and then run under `gdb` and give us a backtrace (`bt`) of where exactly it hangs? I can think of the `accumulate_scans` function erroring and us running into the deadlock described by @MikeGitb in https://github.com/scanse/sweep-sdk/pull/80#discussion_r115117015 - basically never pushing scans into the queue.

> The second issue occurs in the same place, however, when I use the setting hz = 5 and sampleRate = 500, the scans works as expected until a rotational acceleration is applied to the sweep. The motor monetarily stops, and seems to reset.

Hm. @dcyoung @kent-williams you added the motor rotation stabilization feature / code paths - do you have any insights here? The motor momentarily pausing or resetting sounds definitely not right.

dcyoung, 2017-05-08:
If the device is torqued too hard, it can disturb the encoding. If the acceleration changes the speed too much, the device may incorrectly interpret a normal plastic encoder tick as the long tick used for synchronization. Typically this happens in quick succession until the motor adjusts back to a constant speed. During this time, it is possible that successive small encoder ticks can all be misinterpreted as sync ticks. This will manifest as rapid successive scans with low sample count, because the scanner thinks it is seeing the sync byte basically back to back.  The jump can be relatively dramatic (~100 scans with 1-10 readings each). The visualizer checks for this and does not present data until a scan seems valid.

We are looking into a firmware fix that will identify these situations (rotational accelerations that disturb the encoding) and avoid transmitting data entirely until the device settles back down.

A disturbance like this could overflow the scan queue quickly, but it shouldn't lock up I don't think. I'll try to experiment with some intentional torquing to see if i can break libsweep in a similar fashion. But given that you say the code gets locked up in the same place without applying any torque, I doubt this is the root cause of the issue.

justin-p-m, 2017-05-09:
@daniel-j-h with the settings of hz = 5 and sampleRate =1000, we checked the call stack (We are building in VS2017), and the code is hanging up on the function the_cond_var.wait(lock) in queue.h line 51. 

Here is the specific call stack for when the hangup occurs.
`std::_Cnd_waitX(_Cnd_internal_imp_t * _Cnd, _Mtx_internal_imp_t * _Mtx) Line 96`	
`std::condition_variable::wait(std::unique_lock<std::mutex> & _Lck) Line 567`	
`sweep::queue::queue<sweep_scan *>::dequeue() Line 51`	
`sweep_device_get_scan(sweep_device * device, sweep_error * * error) Line 240`

We may have resolved the issue when sampleRate = 500. We double checked our pre-processor definitions and found an error. After we fixed the error, the code shows no signs of locking up when sampleRate = 500.

@dcyoung We did a little bit of testing this morning by applying torque to the sweep for about 30 seconds with the settings of sampleRate = 500, and hz = 5. The motor would stop during the torquing, then attempt to adjust back to a constant speed. During the torquing, the scan queue did seem to overflow but no lock up occurred.

MikeGitb, 2017-05-09:
Most likely the worker thread died for some reason. 
If you have the latest version (as in today) you could set a breakpoint inside of the catch clause of `void sweep_device_accumulate_scans(sweep_device_s device)`  to verify this assumption or even better configure visual studio in such a way that it breaks on exceptions automatically. 
For older versions there is an according error handling block.

justin-p-m, 2017-05-10:
@MikeGitb I think you are right, an exception is being thrown within
`void sweep_device_accumulate_scans(sweep_device_s device)` thread.
The exception is causing the thread to break which in turn means the queue is not being filled, therefore causing the issue we are seeing where the program is getting locked in `the_cond_var.wait(lock)` in queue.h line 51.
Here is the specific call stack when the exception is being thrown within 
`void sweep_device_accumulate_scans(sweep_device_s device)`:

`sweep::serial::device_read(sweep::serial::device * serial, void * to, int len) Line 214`
`sweep::protocol::read_response_scan(sweep::serial::device * serial, sweep::protocol::response_scan_packet_s *scan) Line 98`
`sweep_device_accumulate_scans(sweep_device * device) Line 204`

Please let us know how you think we can resolve this issue.

MikeGitb, 2017-05-10:
@justin-p-m: Unfortunately I don't know what the root cause of the problem is so I can't really help you. For some reason communication between the LIDAR and the Host seems to fail, but I don't have a device myself to test it. It might be helpfull for the developers if you also provide the error message stored in the exception or the line, where the exception is thrown.

I had, however, created a preliminary pull request (#87) that might allow you to work around the problem to some degree:
With this change, `sweep_device_get_scan` will no longer block indefinetely if the worker thread dies but should return an error instead. You could then e.g. reset the device and continue. Obviously that is not a satisfactory solution, but it might help you in the mean time. 

Please keep in mind however, that I deliberately closed that PR, because it is only a halve baked solution and I could not test it myself. Maybe I've time over the weekend to come up with a proper PR that might be more helpfull, but in the end the actual error might be on the level of the serial interface and I'm not familiar with how that works on windows.

justin-p-m, 2017-05-11:
@MikeGitb No worries, I appreciate the help. 
I implemented the work around you mentioned and the fix is working for a little bit until an exception is thrown within protocal.cc. You may be right about the error being on the level of the serial interface. Maybe @daniel-j-h can speak more to this. 

I have included the error message that is stored the in thrown exception:

> Exception thrown at 0x75F1C54F in Project.exe: Microsoft C++ exception: sweep::protocol::error at memory location 0x021A89D0.

I also included the callstack:

> - Project.exe!sweep::protocol::read_response_scan(sweep::serial::device * serial, sweep::protocol::response_scan_packet_s * scan) Line 103
> - Project.exe!sweep_device_accumulate_scans(sweep_device * device) Line 203
> - Project.exe!std::_Invoker_functor::_Call<void (__cdecl*)(sweep_device *),sweep_device *>(void(*)(sweep_device *) && _Obj, sweep_device * && <_Args_0>) Line 1534

@daniel-j-h Please let me know if the call stack is not useful. That way I can work towards getting you a 
backtrace via `gdb`.

Thanks!

daniel-j-h, 2017-05-13:
> I have included the error message that is stored the in thrown exception:
>>  Exception thrown at 0x75F1C54F in Project.exe: Microsoft C++ exception: sweep::protocol::error at memory location 0x021A89D0.

Can you print the exception's message as in

    std::cout << "Error: " << e.what() << std::endl;

It seems like `read_response_scan` fails for some reasons (and I guess that's basically what the exception will tell you then, too).

@dcyoung any idea why responses do not make it through? The response scan reading happens before any sync or error checking. How is it possible that responses don't make it through after we started the scanning process?

I think https://github.com/scanse/sweep-sdk/issues/72 points at the same behavior.

---

> We are looking into a firmware fix that will identify these situations (rotational accelerations that disturb the encoding) and avoid transmitting data entirely until the device settles back down.

@dcyoung any updates on your side?

justin-p-m, 2017-05-16:
@daniel-j-h Sure thing, the error that is printing when the exception is thrown is below:
`Error: invalid response header checksum`

Currently we have been working toward a quick fix by avoiding the worker thread by using the functions
`sweep_device_attempt_start_scanning();` and `sweep_device_get_scan_direct();`
We are running into the same error. Also, just to note we are not using pins 2 and 3 on the connector when using the 2 functions mentioned above.

Please let me know if you need anything else.
Thanks!

dcyoung, 2017-05-17:
> any idea why responses do not make it through?

Hmmm. We've never seen a device fail to send a proper response. Usually when issues like this appear, sniffing the wire with a protocol analyzer shows the expected communication over the wire. The issue is almost always on the host side: serial-usb converter, driver, OS, program, etc.... 

> any updates on your side?

We'll likely release a fix in an upcoming firmware, but there are a few other fixes and priorities that will delay this for a bit. While the data output under angular acceleration is an issue in and of itself, it may not be relevant to the behavior discussed here. I'm working to replicate the described behavior and narrow in on the cause. 

> the error that is printing when the exception is thrown is below: `Error: invalid response header checksum`

Can you print out the contents of the data block just before the checksum is performed.  I'm guessing the contents are not corrupted, but perhaps they have shifted somehow  (program failed to flush/parse etc). In that case, the checksum would fail because it is looking the wrong set of bytes. 

> Also, just to note we are not using pins 2 and 3 on the connector 

Can you elaborate on that? Are you using the provided usb-serial converter? You mentioned you were running windows 7. Have you manually installed the driver, and adjusted the port settings as noted in the [quickstart guide](https://support.scanse.io/hc/en-us/articles/224449728-Quickstart)?

justin-p-m, 2017-05-18:
We are running windows 7 and are using the usb-serial converter. 
The driver suggestion seems to have fixed the issue. Thank you for suggesting that.
I will close the issue. Seems to have been an error on our side. Sorry for the trouble. 

Thanks for the assistance!
Justin

closed 2017-05-18 by justin-p-m