scanse/sweep-sdk #88: RuntimeError: unable to start scanning. could not verify motor speed.

brusangel, 2017-05-10:
#### sweep firmware version
Version 1.3

#### libsweep version + affected bindings

*latest at this moment (20160510- just cloned and installed from git repository)
*working with sweeppy

#### operating system

> Linux raspberrypi 4.4.50-v7+ #970

#### Platform/Hardware Setup

Raspberry pi 3

#### Description:
```
Traceback (most recent call last):
  File "lidar360.py", line 196, in <module>
    main()
  File "lidar360.py", line 154, in main
    sweep.start_scanning()
  File "build/bdist.linux-armv7l/egg/sweeppy/__init__.py", line 136, in start_scanning
RuntimeError: unable to start scanning. could not verify motor speed.

```

dcyoung, 2017-05-10:
Does it `start_scanning()` fail almost immediately, or does it seem to block for ~10 seconds before it fails?
 
Does your code do successfully call any other methods before `start_scanning()` or is `start_scanning()` the first method after device creation?

If there is nothing called before `start_scanning()`... by that I mean your code looks like this:
```bash
with Sweep(dev) as sweep:
        # call start_scanning before any other sweep methods
        sweep.start_scanning()
        ....
```

then this might be related to #73... in which case try the `stop_scanning_buffer_fix` branch.
```bash
git clone https://github.com/scanse/sweep-sdk
git checkout stop_scanning_buffer_fix
# normal install procedure
...
```

brusangel, 2017-05-11:
it fails almost immediately.

`start_scanning()` is the first method called after resource allocation.

I'm applying the "buffer_fix" right now... I'll come back after test. 

thanks!

dcyoung, 2017-05-11:
Woops... the underscores should be dashes. Try  `git checkout stop-scanning-buffer-fix`

brusangel, 2017-05-11:
problem solved

thanks!

brusangel, 2017-05-11:
issue solved/closed!

thank you very much!

closed 2017-05-11 by brusangel