scanse/sweep-sdk #96: sweeppy can't find C-libraries

CodingJonas, 2017-05-22:
#### libsweep version + affected bindings
libsweep 1.1.1

#### operating system
Arch Linux

#### Description:
The make file for the c code installs the libraries into '/usr/local/lib'. The python wrapper 'sweppy' uses those libraries, but looks only in '/usr/lib'.

I fixed it by running
`sudo ln -s /usr/local/lib/libsweep.so /usr/lib/libsweep.so`
`sudo ln -s /usr/local/lib/libsweep.so.1 /usr/lib/libsweep.so.1`

Would be nice if this could work out of the box. Or did I just understand something wrong?

daniel-j-h, 2017-05-22:
Your loader probably does not look into `/usr/local/lib` by default. Or your loader's cache is not up to date. Try the following: remove your symlinks again and confirm it's failing as before.

Then do `sudo ldconfig` to update your loader's cache and try again. If this doesn't work a workaround is to either add `/usr/local/lib` to your runtime's search paths or

    export LD_LIBRARY_PATH="/usr/local/lib/"

dynamically.

On the binding side I think we're doing the correct thing here:
- https://github.com/scanse/sweep-sdk/blob/88f897c95fc1ae346bf504255d947f7c81f0aeb2/sweeppy/sweeppy/__init__.py#L5
- https://docs.python.org/3/library/ctypes.html#ctypes.LibraryLoader.LoadLibrary

daniel-j-h, 2017-06-10:
@CodingJonas any updates here?

daniel-j-h, 2017-08-11:
Stale. Make sure your loader looks for the right paths as described above.

closed 2017-08-11 by daniel-j-h