scanse/sweep-sdk #100: Sweeppy hangs after returning 1~3 scan results

kimauk, 2017-05-26:
#### Description:
I am currently programming python3 application that print Sweep sensor data over terminal for recording.

While doing this, I am having a trouble to print out the data continuously.

When running my code, it stopped printing data and got frozen (?) after printing 1~3 scan results.

I tried both for C++ and Python3

Note that I putted underbars with intention to add spaces in front of a line for better view.

C++ code snippet

while(true){
_   Preformatted textconst sweep::scan scan = device.getscan();
_   for(const sweep::sample& sample : scan.samples){
_   std.cout << “a=” << sample.angle << "d=" << sample.distance << "s=" << sample.signal_strength << "\n";
_   }
}

Python 3 code snippet

with Sweep(dev) as sweep:
_   sweep.startscanning():
_   for scan in sweep.getscans():
_      Print(scan)

#### sweep firmware version
1.4

#### libsweep version + affected bindings
libsweep version: 32bit compiled version using the most recent one from Github (downloaded on 25th May)

specific language binding: Python3 and C++

#### operating system
Windows 10 64bit

Python 3.6.1

dcyoung, 2017-05-29:
Hi there kimauk, 

> Note that I putted underbars with intention to add spaces in front of a line for better view.

When including snippets, you can format your code with markdown. See [here](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code). Could you edit your original post to reformat those snippets, as currently it looks like the indentation in the python could could be problematic. I'm assuming this is just a formatting issue and not indicative of the indentation in your actual code.

Were you able to build and run the examples (`example-c` or `example-c++`) included in `libsweep/examples/`? If not could you try doing so, and report back. Instructions [here](https://github.com/scanse/sweep-sdk/tree/master/libsweep/examples).

What about the example included with sweeppy? Did that work? 

> it stopped printing data and got frozen (?) after printing 1~3 scan results

Did this occur when running the libsweep and sweeppy examples mentioned above? If so could you try building both `libsweep` and the libsweep examples with debug information (`--config Debug`) and then use a debugger to see where exactly the application hangs when executing a libsweep example?

> libsweep version: 32bit compiled version

Is there a reason you're compiling a 32 bit version instead of 64-bit? Is your python installation 32bit or 64 bit? Most likely it is not related, as you are able to print out a few scans. But might try compiling the 64 bit version just to rule out the possibility.

kimauk, 2017-05-30:
I try to build and the the example at the directory lisweep/examples, but could not build and CMake Eorror says LIBSWEEP_INCLUDE_DIR is missing. 

Interestingly, when I typed "echo %LIBSWEEP_INCLUDE_DIR%" it correctly returns the directory, which is "C:\Program Files\sweep\include\sweep". 

Please refer the below attached image. 

[image]

Although I could build and run the libsweep example, it seems the libsweep is well deployed at the directory where it should be, since the python example "python -m sweepy.py COM5" return the data. 

However, after reboot windows, it does not work again. Note that for this I used 64bit libsweep ans 64bit python

The code, which is simply copied  and edit to print data continuously, does not work again after reboot. 

```python
import itertools
import sys
import time
from sweeppy import Sweep

current_milli_time = lambda: int(round(time.time() * 1000))

def main():
    if len(sys.argv) < 2:
        sys.exit('python -m sweeppy <Port name>')

    dev = sys.argv[1]
    with Sweep(dev) as sweep:
        f = open('lidar_output_'+str(current_milli_time())+'txt','a')
        
        speed = sweep.get_motor_speed()
        rate = sweep.get_sample_rate()

        print('Motor Speed: {} Hz'.format(speed))
        print('Sample Rate: {} Hz'.format(rate))

        # Starts scanning as soon as the motor is ready
        try:
            sweep.start_scanning()
            for scan in sweep.get_scans():
                line = str(current_milli_time())+"\n"
                line += '{}\n'.format(scan)
                print(line)
                f.write(line)
        except KeyboardInterrupt:
            pass
        sweep.stop_scanning()
    print('END')
main()
'''

below is the debug results.

![popup_windows](https://cloud.githubusercontent.com/assets/3027049/26580881/05089104-4575-11e7-91fc-7bd03bc9f8c3.PNG)

![cmd output](https://cloud.githubusercontent.com/assets/3027049/26580886/084c34f6-4575-11e7-9a87-04168c2241c4.PNG)

```serial.cc linenumber from 207
  // Check that the number of bytes actually read is equal to the requested quantity
  if (f_result == true)
    f_result = (dw_num_read == (DWORD)len);

  SWEEP_ASSERT(f_result && "reliable read failed to read requested number of bytes");
  if (!f_result)
    throw error{"reading from serial device failed"};
}
'''

agutif, 2017-09-29:
Hi @kimauk,

> I try to build and the the example at the directory lisweep/examples, but could not build and CMake Eorror says LIBSWEEP_INCLUDE_DIR is missing.

> Interestingly, when I typed "echo %LIBSWEEP_INCLUDE_DIR%" it correctly returns the directory, which is "C:\Program Files\sweep\include\sweep".

You have to add this two paths to environment variable:

`C:\Program Files\sweep\lib`
`C:\Program Files\sweep\include`

I know that in this [README.md](https://github.com/scanse/sweep-sdk/tree/master/libsweep) appears that the path is C:\Program Files\sweep\include\sweep but it must be a mistake.

After add this paths to environment variable you have to restart the computer and then follow the instructions [here](https://github.com/scanse/sweep-sdk/tree/master/libsweep/examples) to compile and run the examples.

The example runs fine with a for loop of 10, the original one, but I changed the for loop to an infinite while loop because I am interested in obtaining data continuously and is when the example crash at the same point where the kimauk's code crashed.

Modified example code snippet:
[image]

And this is the error message:
[image]

Thanks in advance.

dcyoung, 2017-09-29:
Hi there @agutif 
Thanks for catching the typo. The REAMDE has now been updated to reflect the proper `PATH` variable additions.

The issue could very well be related to the port settings. Can you double check that you have properly adjusted the port settings as described in this article: https://support.scanse.io/hc/en-us/articles/115000793208-Changing-the-USB-Adapter-Latency-Timer-and-Byte-Size-Setting-on-Windows

agutif, 2017-10-02:
Hi @dcyoung 

Thanks a lot for the info. After adjusting the port settings as described in the article all works fine in an infinite loop.

Thanks again.

dcyoung, 2017-10-02:
Glad that worked for you! Closing this now.

closed 2017-10-02 by dcyoung