scanse/sweep-sdk #111: sweep red light, no data

brusangel, 2017-07-06:
#### sweep firmware version

**1.4**
#### libsweep version + affected bindings

using the Sweep Visualizer, and the Python Examples provided by the sweep-sdk

#### operating system

Debian Jessie (Python 2.7), Visualizer on Windows 8.1 x64

#### Description:

when trying to request the scan data the sweep throw the red light... I went to the documentation and says the red light means a "communication error" and gets no data or partial scans with random data (makes no sense)... I attach a video that shows the problem (please also notice the tick-tick-tick-tick sound the sweep makes)

[https://youtu.be/uyDB04QgT9s](url)

kent-williams, 2017-07-06:
@brusangel - This is a hardware failure. Please contact us at support.scanse.io and we'll get this handled ASAP.

-Kent

closed 2017-07-06 by kent-williams

brusangel, 2017-07-06:
@kent-williams thanks! I just sent a support request there. thanks