scanse/sweep-sdk #112: macOS - Package ´openal´, required by ´sfml-audio´ not found

salman451, 2017-07-12:
#### sweep firmware version
1.4

#### libsweep version + affected bindings
 libsweep.1.1.1.dylib - I cloned sweep-sdk-master.
 It concerns /libsweep/examples/, specifically viewer.cc

#### operating system
 macOS Sierra 10.12.5

#### Platform/Hardware Setup
MacBook Pro 12,5 model (2015)

#### Description:
 The problem arises when I am building the /libsweep/examples/ in macOS terminal. The command that I enter is: 

_cmake .. -DCMAKE_BUILD_TYPE=Release_ 

and the terminal output which follows shows that the SFML2 is not loaded as shown below:

_-- Checking for module 'sfml-all'
--   Package 'openal', required by 'sfml-audio', not found
-- SFML2 required for real-time viewer (libsfml-dev)
-- Checking for module 'libzmq'
--   Found libzmq, version 4.2.2
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/Salman/Documents/sweep-sdk-master/libsweep/examples/build_

Because of this missing reference to openal sfml-all module is not loaded and hence at the next command _cmake --build ._ does not create the exe file for viewer.cc

It cannot find the ´openal´ package. I tried by downloading  the release build of SFML and also compiling the source of SFML to make a release build (it finds openal.framework as it is a native macOS library) and it compiles without any issues. 

I am wondering if this is a bug in libsweep as apparently it cannot find ´openal´ which as I said is a native system library/framework in macOS.

Regards,
Salman.

daniel-j-h, 2017-07-12:
We simply dispatch to pkg-config for finding the sfml libs on your system:

https://github.com/scanse/sweep-sdk/blob/master/libsweep/examples/CMakeLists.txt#L31-L33

I don't have much experience with macOS but I guess you could use brew to install it.

You could also try to change the `sfml-all` pkg-config call to something like:

    sfml-system sfml-window sfml-graphics

or remove the pkg-config call and add the inlucde path and the libs manually here: 

https://github.com/scanse/sweep-sdk/blob/master/libsweep/examples/CMakeLists.txt#L37-L38

salman451, 2017-07-14:
compiles with the above three sfml libraries. However, when i run the example-viewer it repeatedly sends the message: 
Cannot fetch event from a worker thread. (OS X restriction)

salman451, 2017-07-14:
As the issue under discussion is closed I will close this one and open a separate one. thanks

closed 2017-07-14 by salman451