scanse/sweep-sdk #113: Mac: fatal error: ´zmq.hpp´ file not found

salman451, 2017-07-14:
#### sweep firmware version
1.4

#### libsweep version + affected bindings
1.2.1
libsweep/examples in c/c++

#### operating system
macOS Sierra 10.12.5

#### Platform/Hardware Setup
MacBook Pro 12,5 (2015)

#### Description:

Trying to run libsweep/examples/net.cc I receive the following error:
_Project_folder_/libsweep/examples/net.cc:11:10: fatal error: 'zmq.hpp' file not found
#include <zmq.hpp>
         ^
1 error generated.

I have zeromq version 4.2.2 installed using $ brew install zmq.

Also when I configure the build using CMake one of the messages is:
Checking for module 'libzmq'
  Found libzmq, version 4.2.2

So i am assuming the library is installed and found correctly. I checked in the /usr/local/include and only a zmq.h header file exists not zmq.hpp. May be I am missing the correct library that should be used to run this example (?!)

daniel-j-h, 2017-07-14:
I think `brew` does not ship the C++ bindings for ZeroMQ, only the low-level C library. If you want to check out the pub-sub example with ZeroMQ and Protobuf you have to install the C++ wrapper header yourself:

- https://github.com/zeromq/cppzmq/blob/master/zmq.hpp
- http://zeromq.org/topics:cppzmq

It may suffice to copy it to `/usr/local/include/` or add it to the compiler include search path.

salman451, 2017-07-14:
I included zmq.hpp in /usr/local/include/ and it worked. thanks.

closed 2017-07-14 by salman451