scanse/sweep-sdk #121: libsweep Windows build failing

mncosta, 2017-09-08:
#### libsweep version + affected bindings
1.2.3

#### operating system
Windows 10 64 bit Architecture, but building to 32bit using Visual Studio 12 2013

#### Description:
Trying to build libsweep on 32bit arch windows. Did the cmake commands but when i try to then build on Visual Studio 12 2013 i get the following errors:
[Pastebin Error List](pastebin.com/qZRfuQmf)

Not sure if I might need any additional linkage on the cmake or what I'm missing... Any help on the build is appreciated.

Yours faithfully

daniel-j-h, 2017-09-24:
Could you try upgrading Visual Studio; we're making use of C++11 in the implementation and I think your version might be too old. You may also need a more up to date CMake version than, so it knows about current Visual Studio compilers.

mncosta, 2018-07-09:
Ended up building on newer Visual Studio with no problem. Thanks for the help.

closed 2018-07-09 by mncosta