scanse/sweep-sdk #122: Laggy values from Sweep with piCamera integration in Raspberry pi 3 b

Biswajit1995, 2017-10-13:
Hey There,

I am having a issue while integrating Sweep code with one of my code that I was using previously.

Actually as I am working with Raspberry pi (Model 3B), I'm more comfortable with Python, and I am using python3 for the development.

My previous code was simple code to take pictures frame by frame from the RaspiCam and use it after some image processing.(Sorry, I can't post the code over here) but it is fairly simple like the below code snippet: 

```
cap = cv2.VideoCapture(camera_index)
cap.set(3, any_resolution)
cap.set(4, any_resolution)
cap.set(5, frame_rate)
```

then under a continuous loop I'm using `cap.read()` to read out frames and passing them through my another set of code to process and use it.

And It was working perfectly fine unless I started to use it with Sweep Lidar code, from the SDK sweeppy I'm using their startup code [(Here)](https://github.com/scanse/sweep-sdk/blob/master/sweeppy/README.md), and since in the Interface scan.get_scans() is returning an Iterable of Sample class which in turn is a class, I'm accessing it through two for loops as mentioned below:

```
from sweeppy import Sweep
with Sweep('port') as sweep:
    sweep.start_scanning()

    for scan in sweep.get_scans():
        for i in scan:
            for j in i:
                print(j.angle/1000, j.distance)
```
and it was perfectly fine without any lag.
The main problem arises when I integrate this piece of code with `cap.read()` something similar to the below one:

```
from sweeppy import Sweep
with Sweep('port') as sweep:
    sweep.start_scanning()

    for scan in sweep.get_scans():
        for i in scan:
            for j in i:
                 flag, frame = cap.read()
                 print(flag)     # To check the camera read is successful or not
                 print(j.angle/1000, j.distance)
```

here, in the output , its only prints True, or false(In case of camera failure), but not printing the j.angle and j.distance values as it was printing before, there is huge lag, means after 1 min 5 values will get printed then again it will print True or False and it goes on like that.

output example:

```
True
True
True
True
True
True
True
True
True
True
True
True
True
(240, 56)
True
(98, 578)
True
(340, 568)
True
(45, 678)
True
True
True
True
True
True
True
True

```
I don't know why it is behaving like that, any help with this would be great.

And one more thing I want to ask is that, is it necessary to use sweeppy SDK with
`with Sweep('dev') as sweep:` part, is there any alternative to this.

Regards
Biswajit

daniel-j-h, 2017-10-28:
It seems like you're trying to grab frames from your camera as fast as possible in sync with every sample. This is way too often if you assume a 10 Hz rotation and hundreds or thousands of readings per one rotation.

What you should do instead is:
- create one thread responsible for gathering scans, store them and add a timestamp
- create one thread responsible for reading frames from your camera, store them and add a timestamp
- merge both queues associating potentially multiple scans to a single frame

---

The `with Sweep() as` context manager is needed to take care of native memory management making sure resources are getting properly released. You could write a wrapper generator around it e.g.

```python
def scans():
    with Sweep() as sweep:
        for scan in sweep.get_scans():
            yield scan
```

which would yield scans infinitely. You could also add a `abort = yield` to the function to make it a proper coroutine and `.send()` it a value if you want to stop.

dcyoung, 2017-10-30:
@daniel-j-h this use case might be relevant to a lot of projects. Your response above is great, and might help a lot of people start using sweeppy for more complex tasks. Perhaps we should add some similar comments to the sweepy README yeah?

daniel-j-h, 2017-11-07:
Hm yeah I can see how that's a common use case. Feel free to copy my comment above into a readme. Or maybe should we build a small example for Python with two threads / processes.

The easiest way could be to use a [Queue](https://docs.python.org/3.5/library/queue.html), spawn two threads. The first thread pushes scans into the queue; the second thread pulls the most recent `n` scans out of the queue and averages the scans.

daniel-j-h, 2017-11-07:
Added an example for a producer-consumer use-case in https://github.com/scanse/sweep-sdk/pull/125.

subashreeraja, 2018-04-12:
@daniel-j-h , I have a doubt. When I run the command: python -m sweeppy /dev/ttyUSB0 , it executes for certain number of times and exits the program. I want to run it continuously. How should I modify the program? Please help me!

daniel-j-h, 2018-04-13:
The example you get with `python -m sweeppy` is the following which gathers only three scans and prints them:

https://github.com/scanse/sweep-sdk/blob/dee693823022a5ebcec4e556496c671493b2f1c6/sweeppy/sweeppy/__main__.py#L23-L24

I'm closing this issue as your problem is different than the original issue. And the original issue is a user-problem and the examples added in https://github.com/scanse/sweep-sdk/pull/125 show how to properly set up a producer-consumer pipeline.

closed 2018-04-13 by daniel-j-h