scanse/sweep-sdk #123: Support for FreeBSD

hyunh90, 2017-10-20:
#### sweep firmware version

v1.4

#### libsweep version + affected bindings

v1.2.3

#### operating system

```
hyunh@rin:~ % uname -a
FreeBSD rin.kagamine.me 11.1-RELEASE-p1 FreeBSD 11.1-RELEASE-p1 #0: Wed Aug  9 11:17:49 UTC 2017     root@amd64-builder.daemonology.net:/usr/obj/usr/src/sys/GENERIC  i386
```
I also tried on Raspberry Pi 2 (FreeBSD 12-CURRENT); same result (failure).

#### Platform/Hardware Setup

N/A

#### Description:

CMake script fails upon OS detection. FYI, the [CMakeOutput.log](https://github.com/scanse/sweep-sdk/files/1400748/CMakeOutput.log) file is attached.

daniel-j-h, 2017-10-20:
Can you add a branch for FreeBSD to the build system [here](https://github.com/scanse/sweep-sdk/blob/17e937b442eee0c8a213b6e080f33d985d85e569/libsweep/CMakeLists.txt#L17-L39) and try again.

In theory the SDK _should_ work but no one tested it yet.

If you test it and it works feel free to make a pull request enabling compilation on FreeBSD.

hyunh90, 2017-10-20:
Got the build part done.
However, I am getting serial errors: `invalid scan response command` and `invalid response header checksum`.

Will dig further and update you with fixes.

hyunh90, 2017-10-26:
It looks like in the middle of the measurement stream, the checksum byte of a data block suddenly being read as \0(0x00), triggering checksum failure. I suspect this being BREAK condition, as in `serial.cc` the IGNBRK flag has been turned off. But frankly, I don't know why this happens, as I haven't done serial programming before.

For now, I'll keep experimenting with `termios` flags.

hyunh90, 2017-10-27:
Solution found; will send PR in a moment.

closed 2017-10-28 by daniel-j-h

hyunh90, 2017-11-07:
Quick followup: I submitted the [request](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=223433) for FreeBSD Ports tree inclusion of libsweep; it will take few months (or years? XD) for it to be reflected.

PR #124 (2017-10-28): Add support for FreeBSD

#### Scope of changes

As explained in the title, this PR adds libsweep support for FreeBSD and fixes #123.

#### Known Limitations

The real-time viewer and pub-sub networking examples were not tested.